reifsnyderb/Apple-II-R0 #20: Not an issue - PCB routing completed

jimbro1000, 2024-01-31:
I think I've finished routing the PCB with all of the updates and changes I've raised but this is now out of sync with your schematics so I'm not going to just raise a pull request

The cosmetics of the board still need work to get all of the labelling correct, likewise placing of mounting holes

See my fork of the repo for renders and pcb files for comparison

jimbro1000, 2024-01-31:
OK - not quite finished, there is an issue with the SOFT5 net. There are two separate nets labelled SOFT5 generated from A2 (pin 8 and pin 11). Other than the gates driving them being on the same chip, they are not otherwise connected. Given that SOFT5 is just always high (inverted GND) I suggest that the net names are changed... the only impact is fixing DRC errors on the PCB, the schematic is otherwise correct

reifsnyderb, 2024-01-31:
Make them SOFT5a and SOFT5b?  Tracking down this issue could take a while.

jimbro1000, 2024-01-31:
That's what I was thinking - it is really quick to isolate the nets in the PCB editor so I should be able to give you a list of the two sides

jimbro1000, 2024-01-31:
I've found modern substitutes for everything except the character rom but it should be simple enough to convert a 27C16 to suit

reifsnyderb, 2024-01-31:
Thank you for all your work on this!  I just downloaded and was looking at your PCB.  It's looking good!

When ready (no rush), post a list and I'll isolate the two SOFT5 nets.

reifsnyderb, 2024-01-31:
I just found and opened up the discussion section.

Are these modern substitutes still being made?

jimbro1000, 2024-01-31:
Other than the roms I think everything is current on digikey and I've tagged components with the digikey part numbers as I've identified them. The roms can be directly substituted with 27c16 EPROMs but the character Rom is problematic by comparison

closed 2024-02-02 by reifsnyderb